bluerobotics/ardusub-gitbook #48: Doc Bug: Pymavlink · ArduSub GitBook

jaxxzer, 2018-12-06:
Conflict between docs and result:

https://discuss.bluerobotics.com/t/converting-bluerov2-to-auv/3859/3

Bug Page: [Pymavlink · ArduSub GitBook](https://www.ardusub.com/developers/pymavlink.html#pymavlink)

patrickelectric, 2018-12-06:
@jaxxzer, just tested and everything is working this this version under linux and mac. The guy is using freeze and that can explain why he's not get the output.

patrickelectric, 2018-12-06:
@Williangalvani 	can you confirm this ?

patrickelectric, 2018-12-06:
Ok, I did a fresh install on windows in this same version, everything is working fine.

closed 2018-12-06 by patrickelectric

jaxxzer, 2018-12-06:
>  in this version

> in this same version

Which versions? Should we be more explicit in the doc or it's ok?

patrickelectric, 2018-12-06:
I'm going to send a PR to make it more explicit to use python interactive shell.