bluerobotics/ardusub-gitbook #50: Doc Bug: Pymavlink · ArduSub GitBook

vshie, 2019-01-02:
DESCRIBE PROBLEM WITH DOCS HERE

Bug Page: [Pymavlink · ArduSub GitBook](https://www.ardusub.com/developers/pymavlink.html)

Change flight mode example provided reads as if it will command vehicle mode to Stabilize - however when executed changes vehicle mode to Depth hold

jaxxzer, 2019-01-02:
Everything is working fine here.

At 192.168.2.2:8088 terminal:
```
pi@raspberrypi:~ $ screen -r mavproxy
```

Run command on surface computer:
```
$ python test.py 
Command ACCEPTED and EXECUTED
```

Appears in mavproxy console:
```
MANUAL> Got MAVLink msg: COMMAND_ACK {command : 11, result : 0}                 
STABILIZE> Mode STABILIZE
```

jaxxzer, 2019-01-02:
Is QGC open while you are doing this? Try closing it.

closed 2019-01-02 by jaxxzer