bluerobotics/ardusub-gitbook #245: pymavlink set_mode ack, wrong constant being used

refeed, 2022-04-14:
On https://www.ardusub.com/developers/pymavlink.html

There's a part that explains how to wait for the ack:
```
while True:
    # Wait for ACK command
    ack_msg = master.recv_match(type='COMMAND_ACK', blocking=True)
    ack_msg = ack_msg.to_dict()

    # Check if command in the same in `set_mode`
    if ack_msg['command'] != mavutil.mavlink.MAVLINK_MSG_ID_SET_MODE:
        continue

    # Print the ACK result !
    print(mavutil.mavlink.enums['MAV_RESULT'][ack_msg['result']].description)
    break
```
But this constant: `MAVLINK_MSG_ID_SET_MODE` seemingly should be `MAV_CMD_DO_SET_MODE` ? Because when I ran the snippet above, it just stuck in that if statement (`    if ack_msg['command'] != mavutil.mavlink.MAVLINK_MSG_ID_SET_MODE:`)

ES-Alexander, 2022-05-01:
> But this constant: `MAVLINK_MSG_ID_SET_MODE` seemingly should be `MAV_CMD_DO_SET_MODE` ?

Yes, I believe that's correct. `SET_MODE` used to be a message but is now a command, and that example hasn't been updated to reflect that. There's a more up to date example [here](https://gist.github.com/ES-Alexander/ee1dd479dd728b7cef1bf936f9114e71#:~:text=def-,set_mode,-(self%2C%20mode)%3A) as part of a [more complete program](https://github.com/bluerobotics/ardusub-gitbook/issues/234), although it requires Python >= 3.8.

refeed, 2022-05-01:
I see, thanks for the explanation! In your example you used two constants
```
            # check if acknowledged MAV_CMD_DO_SET_MODE or SET_MODE (old)
            if ack_msg.command in (176, 11):
```
I just pushed a PR, should I update it to also put the two constants there?

ES-Alexander, 2022-05-01:
> I just pushed a PR, should I update it to also put the two constants there?

I included both at the time for compatibility, and because I wasn’t sure whether the message was used by any of the send options. Looking again the acknowledgement is using the command protocol, which presumably wasn’t used when it was just a ‘message’, so receiving the message ID there is likely impossible anyway.

closed 2022-05-02 by ES-Alexander