bluerobotics/ardusub-gitbook #253: [ArduSub Documentation]  Full Parameter List · GitBook

cbusse1, 2022-12-07:
### Description
It is unclear from the GitBook whether the documentation is up to date with the **latest stable ArduSub Firmware Version (4.1.0),** see: https://bluerobotics.com/learn/technical-reference/#software-and-firmware

On GitHub you can find that it is indeed not up to date:
[image]

- This might be related to the fact, that the firmware documentation page is not up to date to: https://www.ardusub.com/introduction/required-software/ardusub-autopilot-firmware.html

### Proposal 
- Add the date when the page was last updated and to which ArduSub version it is up to date
- Competing documentation can be somewhat confusing for the user. Maybe it is advised make a note on each outdated GitBook documentation, and provide reference to the more up to date documentation, e.g. for software version refer to: https://bluerobotics.com/learn/technical-reference/#software-and-firmware

### Reference
Bug Page: [Full Parameter List · GitBook](https://www.ardusub.com/developers/full-parameter-list.html)

ES-Alexander, 2022-12-18:
Hi @cbusse1,

In general the documentation in this repo (and currently on ardusub.com) is stale and not being actively updated, while [we're transferring to our new docs system](https://github.com/bluerobotics/ardusub-gitbook/issues/230). I was hoping to have more of the transfer done by now, but it's ended up more spread out than intended.

Agreed that we should probably put notices on the current outdated pages, and links to the new ones if and when they're available. I've just added a link to the new docs in the technical reference table. For the parameters specifically, I've just recently generated those for 4.1, and they're available at https://docs.bluerobotics.com/ardusub-zola/software/autopilot/ArduSub-4.1/developers/parameters/.

ES-Alexander, 2025-04-07:
Both pages now redirect to the new new docs location (😅), [in the ArduPilot wiki](https://ardupilot.org/sub/docs/parameters.html), which includes a version selector for the parameters :-)

closed 2025-04-07 by ES-Alexander